sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library #9: How to call CCLK function

JimKruk, 2022-01-03:
I bought this board SparkFun LTE GNSS Breakout - SARA-R5 GPS-18031, and I am trying to get the clock information (CCLK). In this file: SparkFun_u-blox_SARA-R5_Arduino_Library.h line 82 I found this line under Supported AT Commands:
const char SARA_R5_COMMAND_CLOCK[] = "+CCLK";   // Real-time clock

In this file: SparkFun_u-blox_SARA-R5_Arduino_Library.cpp around line 748 I see a clock function, "String SARA_R5::clock(void)" but cannot figure how to make the call using Arduino code and was wondering if someone could provide some sample code how to call the clock function and get the time back ?
thanks

PaulZC, 2022-01-03:
Hi Jim ( @JimKruk ),

Thank you for raising this issue.

I've added a [new example](https://github.com/sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library/blob/main/examples/SARA-R5_Example11_Clock/SARA-R5_Example11_Clock.ino) to the library showing how to use the ```clock``` functions. While testing it, I found that both ```clock``` functions had minor errors that were stopping them from working correctly... I've corrected the errors in v1.0.4.

Also, please note that the SARA-R5's real-time clock is set during network registration **but only if** your network supports NITZ (Network Identity and Time Zone). If the time comes back as Jan 1st 2015, it means the rtc hasn't been set.

It is possible to set the real-time clock manually but the library does not currently support that. If that's a feature you need, please let me know.

Best wishes,
Paul

JimKruk, 2022-01-03:
Hi Paul
thanks for the fast reply ! IT WORKS !  THANK YOU SO MUCH !!!
Jim

PaulZC, 2022-01-03:
Hi Jim,

Not a problem. Glad that's working for you.

Closing...

Best wishes,
Paul

closed 2022-01-03 by PaulZC