sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library #11: SARA-R5_Example5_ReceiveSMS how to delete old messages ?

JimKruk, 2022-01-06:
I am trying to understand how the receive message code works in example 5. It looks like it executes a function called  mySARA.getPreferredMessageStorage and then calls mySARA.readSMSmessage for each message. Over several days of testing it looks like txt messages received never get deleted and I am worried it will fill up where they are stored (local SIM?)

Is there a 'delete all' messages function I can call once a day ?  I use this on my boat to send txt messages to turn on/off deck lights, get status, etc, but I think I need to delete a message once it is received or just delete all messages periodically.

I was searching and found this command in the ublox doc +CMGD to delete one or more messages but cannot find a function to call to execute it.

thanks !

part of SARA-R5_Example5_ReceiveSMS:

if (mySARA.getPreferredMessageStorage(&used, &total) != SARA_R5_SUCCESS)
  {
    Serial.println(F("An error occurred when trying to read ME memory!"));
  }
  else
  {
    if (used > previousUsed) // Has a new message arrived?
    {
      Serial.print(F("Number of used memory locations: "));
      Serial.println(used);
      Serial.print(F("Total number of memory locations: "));
      Serial.println(total);
      Serial.println();

      int memoryLocation = 1;
      int foundMessages = 0;
      // Keep reading until we find all the messages or we reach the end of the memory
      while ((foundMessages < used) && (memoryLocation <= total))
      {
        String unread = "";
        String from = "";
        String dateTime = "";
        String message = "";
        // Read the message from this location. Reading from empty message locations returns an ERROR
        if (mySARA.readSMSmessage(memoryLocation, &unread, &from, &dateTime, &message) == SARA_R5_SUCCESS)

PaulZC, 2022-01-07:
Hi Jim,

Thanks for raising this issue!

Correct - support for deleting SMS messages was missing. I've added new functions for that in v1.0.5 - which will be released in a few minutes. Please have a look at the new version of Example5 for more details.

Best wishes,
Paul

closed 2022-01-07 by PaulZC

JimKruk, 2022-01-07:
Hi Paul
thanks again for the fast response and excellent customer service !  As you can tell I am not a highly experienced programmer and your great examples are going to enable me to use this hardware and software on my boat to control various DC and AC circuits.  I had been using a 3G solution but here in the USA 3G is being deprecated and this board from SF fills a gap I had being able to send and receive SMS messages.  THANKS AGAIN !