sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library #29: not enough memory for _saraRXBuffer on Atmega32u4 and SARA R500S

leonardojc, 2023-04-17:
### Subject of the issue
I have a custom circuit with a SARA R500S module and an Atmega32u4 processor.
The code compiles well and uploads correctly to the processor, but in the Serial monitor, at the beginning I have the message "Atmega32u4 and SARA R500S not enough memory for _saraRXBuffer" and I can't communicate with the SARA module.

I have tried a Serial Passthrough code, and the M-Center, and everything works correctly.

### Your workbench
* Atmega32u4 + Sara R500S
* Arduino Ide 1.8.15

### Steps to reproduce
* Using any sample code from the library,
* enabling the line mySARA.enableDebugging();

PaulZC, 2023-04-17:
Hi Leonardo (@leonardojc ),

The default size for the ```_saraRXBuffer``` is 2056 bytes (2048 + 8) because the SARA supports AT command lines of up to 2048 characters. The ```_pruneBuffer``` is also 2056 characters. So that's 4K total and I believe the ATmega32U4 only has 2.5KB of SRAM available...

Please try changing the ```_RXBuffSize``` definition to say 520 bytes (512 + 8). That _may_ allow the examples to run, but you will at some point run into problems due to the buffers not being large enough.

https://github.com/sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library/blob/2b0d5b523b258093b64737b37d2a1a5c690532ad/src/SparkFun_u-blox_SARA-R5_Arduino_Library.h#L965

The best solution is to choose a processor with more RAM...

Best wishes,
Paul

closed 2023-04-17 by PaulZC

PaulZC, 2023-04-17:
OK. Please try 264 (256 + 8). It _may_ work?

Best wishes,
Paul

leonardojc, 2023-04-17:
no luck
with 264 gives error "not enough memory for _saraResponseBacklog!"
241 is the highest value I can put without errors, but there is no response from the module, which is connected to HW Serial1 and works with the other Serial Passthrough code

serial monitor output
```
sendCommandWithResponse: Command: 
init: Begin module init.
init: Attempting autobaud connection to module.
sendCommandWithResponse: Command: 
sendCommandWithResponse: Command: 
sendCommandWithResponse: Command: 
sendCommandWithResponse: Command: 
sendCommandWithResponse: Command: 
sendCommandWithResponse: Command: 
init: Begin module init.
init: Power cycling module.

```

leonardojc, 2023-04-17:
Following the logic I put the value in 136 (128+8) and I get this response

```
init: Begin module init.
sendCommandWithResponse: Command: E0
init: Module failed echo test.
init: Begin module init.
init: Attempting autobaud connection to module.
sendCommandWithResponse: Command: +IPR=115200
sendCommandWithResponse: Command: 
sendCommandWithResponse: Command: +IPR=115200
sendCommandWithResponse: Command: 
sendCommandWithResponse: Command: +IPR=115200
sendCommandWithResponse: Command: 
sendCommandWithResponse: Command: +IPR=115200
sendCommandWithResponse: Command: 
sendCommandWithResponse: Command: +IPR=115200
sendCommandWithResponse: Command: 
sendCommandWithResponse: Command: +IPR=115200
sendCommandWithResponse: Command: 
init: Begin module init.
init: Power cycling module.

```

PaulZC, 2023-04-18:
Hi Leonardo,

The best solution is to use a processor that has more RAM. But, right now, I can't explain why it is not at least initializing with _RXBuffSize set to 128. I will try and investigate this if I have some spare time - but I have a lot of other projects to complete first.

Best wishes,
Paul

leonardojc, 2023-04-18:
Don't worry Paul, thank you very much it was a great help