sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library #33: ARA-R500S and AZURE DPS

leonardojc, 2023-09-14:
I need to connect SARA-R500S modules to an AZURE DPS service.
using CA, CC and PK certificates.
The Sara Module is connected to an ESP32 via UART and AT commands

Is there any information or example on how to use this library to perform this task?

thank you so much

PaulZC, 2023-09-14:
Hi Leonardo,

Sorry, no, we don't have an example for that. But the library does support +USECMNG via the ```setSecurityManager``` method:

https://github.com/sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library/blob/d8adcb3be61f3c658caa25b6896dc537da6b568e/src/SparkFun_u-blox_SARA-R5_Arduino_Library.cpp#L4789

For more details, please see +USECMNG in the [AT Commands Manual](https://www.u-blox.com/sites/default/files/SARA-R5_ATCommands_UBX-19047455.pdf). The enums for the op_code and security type are defined here:

https://github.com/sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library/blob/d8adcb3be61f3c658caa25b6896dc537da6b568e/src/SparkFun_u-blox_SARA-R5_Arduino_Library.h#L610-L621

If you get this working, please submit a new example as a Pull Request. We really appreciate it.

I will leave this issue open for now. Please feel free to ask for advice or suggestions here and I will help if I can.

Very best wishes,
Paul

leonardojc, 2023-09-14:
Hi Paul, thanks for your response.
I have tried to connect manually with AT commands, but I have had no luck, on the contrary I can connect without problems to AWS IOT services, using X.509 certificates
but with Azure, I don't know where to put the DSP's SCOPE ID, which is mandatory for the connection.

PaulZC, 2023-09-14:
Hi,

Which version of the SARA-R5 are you using? Is it the 'special' AWS IoT ExpressLink version? That version is - I think - limited to AWS...

Best,
Paul

leonardojc, 2023-09-14:
No, it's a custom board with an ESP32 and a SARA-R500S-01B-00

PaulZC, 2023-09-14:
OK - thanks.

I'd recommend asking your question in the u-blox support forum. Someone there may know how to set the SCOPE ID. If you don't get a reply, let me know and I will pass your question along to a u-blox Field Application Engineer.

Best wishes,
Paul

PaulZC, 2023-09-14:
Link: https://portal.u-blox.com/s/topic/0TO2p000000Hr7cGAC/cellular

I'd recommend asking a Public Question first, then escalate to a Support Case if needed.

leonardojc, 2023-09-14:
Thank you very much Paul, I have already sent a question to u-blox support, I am waiting for a response.
I keep you informed,
Best wishes,
Leonardo

leonardojc, 2023-09-20:
> OK - thanks.
> 
> I'd recommend asking your question in the u-blox support forum. Someone there may know how to set the SCOPE ID. If you don't get a reply, let me know and I will pass your question along to a u-blox Field Application Engineer.
> 
> Best wishes, Paul

Hi Paul. I have consulted in the forum, and I have also sent a message to support, but I have not received any response.
If you have contact with a U-Blox engineer, it would be very helpful.
It is not a simple project, it is a prototype for a device that is going to be mass produced.
but I'm stuck trying to connect to Azure.

best

Leo

leonardojc, 2023-09-27:
hi @PaulZC , I don't want to bother you, but I haven't gotten any answers from anywhere.
Can you get a contact from someone at Ublox?
I will really appreciate your help
best 
Leo

PaulZC, 2023-09-27:
Hi Leo,
I passed your question along to two u-blox FAE's, but they are busy with a big conference in Las Vegas this week. They will answer but it may take them a few more days.
Apologies for the delay,
Paul

leonardojc, 2023-09-28:
> Hi Leo, I passed your question along to two u-blox FAE's, but they are busy with a big conference in Las Vegas this week. They will answer but it may take them a few more days. Apologies for the delay, Paul

I can’t thank you enough!

Leo

PaulZC, 2023-10-13:
Hi Leo (@leonardojc ),

We got a helpful reply from the u-blox FAE. He has not had any hands-on experience with setting up connections to Azure, but he has researched the issue and hopes the following will help:

A number of the Azure examples use the Azure SDK and don’t show how ID scope is used, but [this doc](https://learn.microsoft.com/en-us/azure/iot/iot-mqtt-connect-to-iot-dps) describes connecting via MQTT instead of using the SDK. This shows that the ID scope is used in the MQTT username field (AT+UMQTT=4 command):

[image]

I hope this helps you with your project.

Very best wishes,
Paul

closed 2023-10-13 by PaulZC