sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library #37: Not properly an issue: understanding when battery must be replaced

Davefries1966, 2024-01-12:
Hi all,

I'm using a Sparkfun AssetTracker development kit with MicroMod ESP32 add-on. The deployed device works in a unmanned environment and communicates via MQTT, as per closed issue https://github.com/sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library/issues/34 (thans Paul!). All is working fine now, I just would implement some diagnostics on the device, to be reported through special MQTT topics  (e.g., overtemperature, exhausted battery and so on...).

I ran the example AssetTracker_Example4_

### Your workbench
* What platform are you using?
* What version of the device are you using? Is there a firmware version?
* How is the device wired to your platform?
* How is everything being powered?
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue to a gist.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

Davefries1966, 2024-01-12:
Hi all,

I'm using a Sparkfun AssetTracker development kit with MicroMod ESP32 add-on. The deployed device works in a unmanned environment and communicates via MQTT, as per closed issue https://github.com/sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library/issues/34 (thans Paul!). All is working fine now, I just would implement some diagnostics on the device, to be reported through special MQTT topics (e.g., overtemperature, exhausted battery and so on...).

I ran the example AssetTracker_Example4_MAX17048 which helps extracting a lot of information about battery status. I don't understand how to interpret the above information to establish exactly when the battery is DEAD and MUST be replaced (usually a good indicator is when charger is ON and the charging current is NEGATIVE).

Thanks for everyone could give me any hints.

David

PaulZC, 2024-01-14:
Hi David,

Please clarify. When you say DEAD do you mean: discharged and requires charging; or that the battery is no longer holding charge?

You should be able to infer either by monitoring the battery voltage, State Of Charge, and Change Rate?

Best wishes,
Paul

Davefries1966, 2024-01-14:
Hi Paul

happy to meet you again here, and thanks for your prompt support ! :-)

I mean DEAD when the battery is no longer holding charge, that is, it is completely exhausted.

I was able to run the example and read all the above information, but it is unclear to me how to match them one to each other.

Cheers
David

PaulZC, 2024-01-15:
Hi David,

There is some useful information in this application note: https://www.analog.com/en/technical-articles/battery-gauge--maxim-integrated.html

I wonder if the following would work? Similar to the stepped-discharge test:
Ensure your system is running from the battery
Use the D0 or D1 breakout pad to control a small external circuit: a transistor or FET low-side switch which connects a 33 Ohm load to 3V3 (adding 100mA to the current draw)
With the external load enabled, check the increased discharge rate matches a known good battery. If the discharge rate is higher, the battery may require replacement?

Or, sample and average the battery voltage with the load turned off. Then repeat with the load turned on. The voltage drop from a failing battery will be higher than a new battery?

I hope this helps,
Paul

Davefries1966, 2024-01-15:
Hi Paul

I will take a look there!

Cheers
David

PaulZC, 2024-02-04:
Hi David,

Closing... Please re-open if you need more help with this.

Best wishes,
Paul

closed 2024-02-04 by PaulZC

Davefries1966, 2024-04-19:
ok thanks Paul ! :-)