sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library #38: %zu not supported by newlib printf Arduino_STM32 toolchain

fronders, 2024-02-04:
### Subject of the issue
Describe your issue here.

### Your workbench
* What platform are you using?
* What version of the device are you using? Is there a firmware version?
* How is the device wired to your platform?
* How is everything being powered?
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue to a gist.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

PaulZC, 2024-02-04:
Hi @fronders ,

Glad you got it working. How would you like to resolve this? I am happy to pin it, to make it easier for others to find and re-use your solution.

Best wishes,
Paul

fronders, 2024-02-05:
Hi @PaulZC,

I think the best would be to update the code to make it usable for everyone, even ARM toolchain with nano version of newlib.
See the PR with some more fixes :)

PR #39 (2024-02-27): Update getFileBlock
This addresses several things:

- fix #38 by changing the format string to use `%lu` specifier which will work for everyone
- add more error checking and command formatting to make `getFileBlock()` similar to other methods
- make more reliable detection of `comma_idx` - the offset of file size in `+URDBLOCK` response