sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library #41: HardwareSerial communication with Nina w106 (esp32 chip)

fcziborr, 2024-04-11:
I am reaching out to you as I have encountered an issue that I cannot solve on my own. I am using the Sara R5 modem for internet communication and interfacing the modem via hardware serial using my ESP32. For this, I have been successfully using the SparkFun library (<SparkFun_u-blox_SARA-R5_Arduino_Library.h>).

However, I have recently integrated the u-blox Nina 106 as a microcontroller (µC), which also utilizes an ESP32 chip. With this device, hardware serial communication is not functioning as expected. While I can successfully control my hardware serial outputs, it is not possible to use the library <SparkFun_u-blox_SARA-R5_Arduino_Library.h> to realize hardware serial communication with the modem. This renders communication with the modem via hardware serial impossible. However, communication via software serial works flawlessly.

It seems that the issue may lie within the library software, as I am able to control the hardware serial outputs of the Nina 106 successfully without this library. Therefore, I suspect it could be a software problem.

* i am using the Nina W106 as µC and the Arduino IDE?
* i am using the current version of the library (04.11.2024)
* my nina w106 is connected via HardwareSerial pins of the w106?
* my device is powerd with a 3400mAh battery

Best regards

PaulZC, 2024-04-11:
Hi @fcziborr ,

Which Arduino IDE Board package are you using?
(Are you using the standard Espressif Arduino ESP32 Boards? Which version?)
Which "board" have you selected?

Can you please post your code where you define your ```HardwareSerial``` port, begin it, and begin the SARA_R5? I suspect it looks something like:

```
// In globals
HardwareSerial mySerial(1); // Use UART1
int8_t myTxPin = ?; // GPIO pin for UART TX
int8_t myRxPin = ?; // GPIO pin for UART RX

// In setup
mySerial.begin(9600, SERIAL_8N1, myRxPin, myTxPin);
if (mySARA.begin(mySerial, 9600) ) {
  Serial.println(F("SARA-R5 connected!"));
}
```

Thanks - and best wishes,
Paul

fcziborr, 2024-04-15:
Many thanks for the quick response.
Here is an example code:

`/*

  SARA-R5 Example
  ===============

  Identification - using ESPSoftwareSerial
  
  Written by: Paul Clark
  Date: December 29th 2021

  This example demonstrates how to read the SARA's:
    Manufacturer identification
    Model identification
    Firmware version identification
    Product Serial No.
    IMEI identification
    IMSI identification
    SIM CCID
    Subscriber number
    Capabilities
    SIM state

  The ESP32 core doesn't include SoftwareSerial. Instead we use the library by Peter Lerup and Dirk O. Kaar:
  https://github.com/plerup/espsoftwareserial

  Feel like supporting open source hardware?
  Buy a board from SparkFun!

  Licence: MIT
  Please see LICENSE.md for full details

*/

//#include <SoftwareSerial.h> //Click here to get the library: http://librarymanager/All#ESPSoftwareSerial_ESP8266/ESP32
#include <HardwareSerial.h> // ist bereits teil von Arduino IDE
#include <SparkFun_u-blox_SARA-R5_Arduino_Library.h> //Click here to get the library: http://librarymanager/All#SparkFun_u-blox_SARA-R5_Arduino_Library

#define DEBUG_PASSTHROUGH_ENABLED

#define RX_Pin          23   //Nina W16 23    Esp32 16                          
#define TX_Pin          18   //Nina W1618    Esp32 17           
#define CTS_Pin         19   //Nina W16 19    Esp32 4
#define RTS_Pin         5    //Nina W16 5     Esp32 0
#define HW_Port          2   
#define Sara_Baud      9600  //9600
#define PWR_Pin         27   //Nina W16 27   Esp32 26          
#define RST_Pin         32   //Nina W16 32   Esp32 27

HardwareSerial saraSerial(HW_Port);

SARA_R5 mySARA(PWR_Pin, RST_Pin);                             // Create a SARA_R5 object Power Pin, Reset Pin

// Map SIM states to more readable strings
String simStateString[] =
{
  "Not present",      // 0
  "PIN needed",       // 1
  "PIN blocked",      // 2
  "PUK blocked",      // 3
  "Not operational",  // 4
  "Restricted",       // 5
  "Operational"       // 6
};

// processSIMstate is provided to the SARA-R5 library via a 
// callback setter -- setSIMstateReadCallback. (See setup())
void processSIMstate(SARA_R5_sim_states_t state)
{
  Serial.println();
  Serial.print(F("SIM state:           "));
  Serial.print(String(state));
  Serial.println();
}

void setup()
{
  Serial.begin(115200); // Start the serial console
  mySARA.enableDebugging();                                   // Uncomment this line to enable helpful debug messages on Serial

  // Wait for user to press key to begin
  Serial.println(F("SARA-R5 Example"));
  Serial.println(F("Press any key to begin"));
  
  while (!Serial.available()) // Wait for the user to press a key (send any serial character)
    ;
  while (Serial.available()) // Empty the serial RX buffer
    Serial.read();

  //mySARA.enableDebugging(); // Uncomment this line to enable helpful debug messages on Serial

  // For the MicroMod Asset Tracker, we need to invert the power pin so it pulls high instead of low
  // Comment the next line if required
  mySARA.invertPowerPin(true);

  saraSerial.begin(Sara_Baud, SERIAL_8N1, RX_Pin, TX_Pin, false);        // Hardware Serial
 // saraSerial.setPins(RX_Pin, TX_Pin, CTS_Pin, RTS_Pin);
 // saraSerial.setHwFlowCtrlMode(HW_FLOWCTRL_CTS_RTS);
 
  saraSerial.write("ATI");    // Test AT command. The command is received at the modem (measured with an oscilloscope)

  // Initialize the SARA
  if (mySARA.begin(saraSerial, Sara_Baud) )
  {
    Serial.println(F("SARA-R5 connected!"));
  }
  else
  {
    Serial.println(F("Unable to communicate with the SARA."));
    Serial.println(F("Manually power-on (hold the SARA On button for 3 seconds) on and try again."));
    while (1) ; // Loop forever on fail
  }
  Serial.println();

  Serial.println("Manufacturer ID:     " + String(mySARA.getManufacturerID()));
  Serial.println("Model ID:            " + String(mySARA.getModelID()));
  Serial.println("Firmware Version:    " + String(mySARA.getFirmwareVersion()));
  Serial.println("Product Serial No.:  " + String(mySARA.getSerialNo()));
  Serial.println("IMEI:                " + String(mySARA.getIMEI()));
  Serial.println("IMSI:                " + String(mySARA.getIMSI()));
  Serial.println("SIM CCID:            " + String(mySARA.getCCID()));
  Serial.println("Subscriber No.:      " + String(mySARA.getSubscriberNo()));
  Serial.println("Capabilities:        " + String(mySARA.getCapabilities()));

  // Set a callback to return the SIM state once requested
  mySARA.setSIMstateReportCallback(&processSIMstate);
  // Now enable SIM state reporting for states 0 to 6 (by setting the reporting mode LSb)
  if (mySARA.setSIMstateReportingMode(1) == SARA_R5_SUCCESS)
    Serial.println("SIM state reports requested...");
  // You can disable the SIM staus reports again by calling assetTracker.setSIMstateReportingMode(0)
}

void loop()
{
  mySARA.poll(); // Keep processing data from the SARA so we can extract the SIM status
}
`

This code works flawlessly on the ESP32. The pins used are for the Nina W106 and the ESP32 are listed with comments.

The command "saraSerial.write("ATI");" is inserted for testing purposes and is successfully received by the modem, as verified by oscilloscope measurements. However, starting from line 103 (if (mySARA.begin(saraSerial, Sara_Baud))), the AT commands are no longer received and cannot be measured.
I am using the ESP32 Wroom 32 with the standard library from Espressif and the ESP32 Dev Module. For the Nina W106, I am also utilizing the standard library from Espressif with the u-blox NINA-W10 series (ESP32) version 2.0.11.

Thanks and best regards
Frederik

PaulZC, 2024-04-15:
Hi Frederik (@fcziborr ),

Ah, OK, I understand what is happening. The library calls ```beginSerial``` which [(re)initializes the serial port](https://github.com/sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library/blob/b0f9e3c23f7467d1c96841bab6c30455e8570982/src/SparkFun_u-blox_SARA-R5_Arduino_Library.cpp#L6627-L6643). In your case it is (probably) overwriting the pin numbers etc. with defaults.

```beginSerial``` is ```virtual```. It can be overridden it in a custom class. Please try:

```
// In globals
class SARA_R5_derived : public SARA_R5 {
protected:
  void beginSerial(unsigned long baud) override
  {
    saraSerial.begin(baud, SERIAL_8N1, RX_Pin, TX_Pin, false); // Hardware Serial
    // saraSerial.setPins(RX_Pin, TX_Pin, CTS_Pin, RTS_Pin);
    // saraSerial.setHwFlowCtrlMode(HW_FLOWCTRL_CTS_RTS);
  }
}
SARA_R5_derived mySARA(PWR_Pin, RST_Pin); // Create a SARA_R5 object Power Pin, Reset Pin
```

The ```saraSerial.begin``` in ```setup``` becomes redundant.

I hope this solves your issue.

Best wishes,
Paul

fcziborr, 2024-04-15:
Thank you so much. I've incorporated the changes, and everything is working  now.
best regards
Frederik

PaulZC, 2024-04-15:
Thank you for the update Frederik - I am glad it is working!

I am going to close this issue. Please re-open if you need more help with this.

Best wishes,
Paul

closed 2024-04-15 by PaulZC

fcziborr, 2024-04-29:
Hello, I need your help again.

When I make the changes in the library it works. However, when I integrate the changes into the program code as you suggested, I encounter the following error message:

"Compilation error: no matching function for call to 'SARA_R5_derived::SARA_R5_derived(int, int)"

If I don't pass the arguments (PWR_Pin, RST_Pin) to the function "SARA_R5_derived mySARA", the code compiles, but the pins are not passed to the library as desired.
Here is my code:
`/*

  SARA-R5 Example
  ===============

  Identification - using ESPSoftwareSerial
  
  Written by: Paul Clark
  Date: December 29th 2021

  This example demonstrates how to read the SARA's:
    Manufacturer identification
    Model identification
    Firmware version identification
    Product Serial No.
    IMEI identification
    IMSI identification
    SIM CCID
    Subscriber number
    Capabilities
    SIM state

  The ESP32 core doesn't include SoftwareSerial. Instead we use the library by Peter Lerup and Dirk O. Kaar:
  https://github.com/plerup/espsoftwareserial

  Feel like supporting open source hardware?
  Buy a board from SparkFun!

  Licence: MIT
  Please see LICENSE.md for full details

*/

//#include <SoftwareSerial.h> //Click here to get the library: http://librarymanager/All#ESPSoftwareSerial_ESP8266/ESP32
#include <HardwareSerial.h> // ist bereits teil von Arduino IDE
#include <SparkFun_u-blox_SARA-R5_Arduino_Library.h> //Click here to get the library: http://librarymanager/All#SparkFun_u-blox_SARA-R5_Arduino_Library

#define DEBUG_PASSTHROUGH_ENABLED

#define RX_Pin          23   //23    16                          
#define TX_Pin          18   //18    17           
#define CTS_Pin         19   //19      4
#define RTS_Pin         5    //5      0
#define HW_Port          2   
#define Sara_Baud      9600  //9600
#define PWR_Pin         27   //27   26          
#define RST_Pin         32   //32   27

HardwareSerial saraSerial(HW_Port);
//SARA_R5 mySARA(PWR_Pin, RST_Pin);                             // Create a SARA_R5 object Power Pin, Reset Pin

// In globals
class SARA_R5_derived : public SARA_R5 {
protected:
  void beginSerial(unsigned long baud) override
  {
    saraSerial.begin(baud, SERIAL_8N1, RX_Pin, TX_Pin, false); // Hardware Serial
    // saraSerial.setPins(RX_Pin, TX_Pin, CTS_Pin, RTS_Pin);
    // saraSerial.setHwFlowCtrlMode(HW_FLOWCTRL_CTS_RTS);
  }
};
SARA_R5_derived mySARA(PWR_Pin, RST_Pin); // Create a SARA_R5 object Power Pin, Reset Pin

// Map SIM states to more readable strings
String simStateString[] =
{
  "Not present",      // 0
  "PIN needed",       // 1
  "PIN blocked",      // 2
  "PUK blocked",      // 3
  "Not operational",  // 4
  "Restricted",       // 5
  "Operational"       // 6
};

// processSIMstate is provided to the SARA-R5 library via a 
// callback setter -- setSIMstateReadCallback. (See setup())
void processSIMstate(SARA_R5_sim_states_t state)
{
  Serial.println();
  Serial.print(F("SIM state:           "));
  Serial.print(String(state));
  Serial.println();
}

void setup()
{
  Serial.begin(115200); // Start the serial console

  // Wait for user to press key to begin
  Serial.println(F("SARA-R5 Example"));
  Serial.println(F("Press any key to begin"));
  
  while (!Serial.available()) // Wait for the user to press a key (send any serial character)
    ;
  while (Serial.available()) // Empty the serial RX buffer
    Serial.read();

  mySARA.enableDebugging(); // Uncomment this line to enable helpful debug messages on Serial

  // For the MicroMod Asset Tracker, we need to invert the power pin so it pulls high instead of low
  // Comment the next line if required
  mySARA.invertPowerPin(true);

  saraSerial.begin(Sara_Baud, SERIAL_8N1, RX_Pin, TX_Pin, false);        // Hardware Serial

  saraSerial.setPins(RX_Pin, TX_Pin, CTS_Pin, RTS_Pin);
  saraSerial.setHwFlowCtrlMode(HW_FLOWCTRL_CTS_RTS);
  saraSerial.write("ATI");

  // Initialize the SARA
  if (mySARA.begin(saraSerial, Sara_Baud) )
  {
    Serial.println(F("SARA-R5 connected!"));
  }
  else
  {
    Serial.println(F("Unable to communicate with the SARA."));
    Serial.println(F("Manually power-on (hold the SARA On button for 3 seconds) on and try again."));
    while (1) ; // Loop forever on fail
  }
  Serial.println();

  Serial.println("Manufacturer ID:     " + String(mySARA.getManufacturerID()));
  Serial.println("Model ID:            " + String(mySARA.getModelID()));
  Serial.println("Firmware Version:    " + String(mySARA.getFirmwareVersion()));
  Serial.println("Product Serial No.:  " + String(mySARA.getSerialNo()));
  Serial.println("IMEI:                " + String(mySARA.getIMEI()));
  Serial.println("IMSI:                " + String(mySARA.getIMSI()));
  Serial.println("SIM CCID:            " + String(mySARA.getCCID()));
  Serial.println("Subscriber No.:      " + String(mySARA.getSubscriberNo()));
  Serial.println("Capabilities:        " + String(mySARA.getCapabilities()));

  // Set a callback to return the SIM state once requested
  mySARA.setSIMstateReportCallback(&processSIMstate);
  // Now enable SIM state reporting for states 0 to 6 (by setting the reporting mode LSb)
  if (mySARA.setSIMstateReportingMode(1) == SARA_R5_SUCCESS)
    Serial.println("SIM state reports requested...");
  // You can disable the SIM staus reports again by calling assetTracker.setSIMstateReportingMode(0)
}

void loop()
{
  mySARA.poll(); // Keep processing data from the SARA so we can extract the SIM status
}
`

PaulZC, 2024-04-29:
Hi Frederik (@fcziborr ),

Please try:

```
#include <HardwareSerial.h>
#include <SparkFun_u-blox_SARA-R5_Arduino_Library.h> //Click here to get the library: http://librarymanager/All#SparkFun_u-blox_SARA-R5_Arduino_Library

#define RX_Pin 23
#define TX_Pin 18
#define CTS_Pin 19
#define RTS_Pin 5
#define HW_Port 2
#define Sara_Baud 9600
#define PWR_Pin 27
#define RST_Pin 32
#define Max_Init 3 // maxInitTries

HardwareSerial saraSerial(HW_Port);

class SARA_R5_derived : public SARA_R5 {
public:
  // constructor
  SARA_R5_derived() : SARA_R5{ PWR_Pin, RST_Pin, Max_Init } {}

protected:
  void beginSerial(unsigned long baud) override
  {
    saraSerial.end();
    saraSerial.begin(baud, SERIAL_8N1, RX_Pin, TX_Pin, false); // Hardware Serial
    saraSerial.setPins(RX_Pin, TX_Pin, CTS_Pin, RTS_Pin);
    saraSerial.setHwFlowCtrlMode(HW_FLOWCTRL_CTS_RTS);
  }
};

SARA_R5_derived mySARA; // Create a SARA_R5_derived object

void setup() {

  Serial.begin(115200);

  // Initialize the SARA
  if (mySARA.begin(saraSerial, Sara_Baud) )
  {
    Serial.println(F("SARA-R5 connected!"));
  }
}

void loop() {
}
```

Best wishes,
Paul

fcziborr, 2024-05-13:
Thank you very much, Paul. I've integrated your changes into the programcode, and it's up and running. 
Best regards, Frederik.