sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library #48: Read Subscribe message

PhilC66, 2024-06-25:
### Subject of the issue
command readMQTT() take 5 seconds to give answer if no message,
take 1second if message present

### Your workbench
* What platform are you using?
* ESP32
* What version of the device are you using? Is there a firmware version?
* SARA-R510M8S 02.06
* How is the device wired to your platform?
* How is everything being powered?
* Are there any additional details that may help us help you?
* use : SARA-R5_Example17_ThingSpeak_MQTT.ino
with empty message
17:18:26.007 -> debut:59236
17:18:26.007 -> sendCommandWithResponse: Command: +UMQTTC=6,1
17:18:26.007 -> sendCommandWithResponse: Response: 
17:18:26.007 -> OK
17:18:31.006 -> 
17:18:31.006 -> readMQTT: sendCommandWithResponse err 4
17:18:31.006 -> fin:64236

with message
17:18:26.007 -> debut:59236
17:18:26.007 -> sendCommandWithResponse: Command: +UMQTTC=6,1
17:18:26.007 -> sendCommandWithResponse: Response: 
17:18:26.007 -> OK
17:18:31.006 -> 
17:18:31.006 -> readMQTT: sendCommandWithResponse err 4
17:18:31.006 -> fin:64236

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue to a gist.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

PaulZC, 2024-06-25:
Hi @PhilC66 ,

I think this is probably a timeout issue?

First up, it looks like you posted the same serial console messages twice? "with empty message" and "with message" are the same - both show a 5 second delay. But I understand the issue.

In ```readMQTT```, ```sendCommandWithResponse``` is called with a timeout of ```(5 * SARA_R5_STANDARD_RESPONSE_TIMEOUT)``` i.e. 5 seconds.

So, I think this is expected. If the topic is 'empty', it will take ```readMQTT``` to timeout.

Closing. Please reopen if you need more help with this.

Best wishes,
Paul

closed 2024-06-25 by PaulZC

PhilC66, 2024-06-25:
0K, I understand. 
How I can avoid polling message instead.
may be using URC?
how to do that?
Thanks for your help
Philippe

PaulZC, 2024-06-25:
Hi Philippe,

Looking at the SARA-R5 AT commands manual, the ```+UUMQTTC:6``` URC indicates how many unread messages are in the buffer.

You will need to modify Example17:

Modify the callback ```processMQTTcommandResult```. If ```command``` == 6 (```SARA_R5_MQTT_COMMAND_READ```), ```result``` will indicate the number of unread messages. Use this to set a global flag.

Keep calling ```bufferedPoll``` in the loop.

Only call ```readMQTT``` when the global flag indicates messages are waiting to be read.

I hope this helps,
Paul

PhilC66, 2024-06-26:
High Paul,
Very good, I didn't know.
I learned something
Thank you so much
Philippe