sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library #49: ESP32 crash if enableDebugging() not enabled

PhilC66, 2024-07-03:
### Subject of the issue
Describe your issue here.
if enableDebugging() not enabled, ESP32 crash when Initialize the SARA mySARA.begin(saraSerial, Sara_Baud)

in SARA_R5_error_t SARA_R5::init(unsigned long baud,
                              SARA_R5::SARA_R5_init_type_t initType)
beginSerial(baud);
_debugPort->print("initype:");
_debugPort->println(initType);

_debugPort is null if enableDebugging() not enabled

needs to add 
if (_printDebug == true) befor 

Thanks

### Your workbench
* What platform are you using?
* What version of the device are you using? Is there a firmware version?
* How is the device wired to your platform?
* How is everything being powered?
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue to a gist.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

PaulZC, 2024-07-03:
Hi Philippe (@PhilC66 ),

Which version of the library do you have installed? Or maybe you modified the library source code? The latest code looks OK to me:

https://github.com/sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library/blob/cc961b2870059eb7a9f91cfc99af18187d614ae2/src/SparkFun_u-blox_SARA-R5_Arduino_Library.cpp#L5820-L5836

Closing... Please reopen if you need more help with this.

Best wishes,
Paul

closed 2024-07-03 by PaulZC

PhilC66, 2024-07-03:
High Paul,
Same version, I modified this version a few days ago, and I forgot
Sorry for the inconvenience
Thanks
Philippe