sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library #50: Sparkfun AssetTracker: MQTT problem when connecting through LTE modem

Davefries1966, 2024-07-19:
Hi all,
As per previous related issues, I purchased a Sparkfun AssetTracker development kit with MicroMod ESP32 add-on, LTE/GNSS antennas and so on.

I developed an application which connects to a MOSQUITTO broker on port 8883 (in TLS/SSL mode) and sends some topics to it.
All works flawlessly when connecting to the MOSQUITTO test broker (test.mosquitto.org:8883) with both wifi and LTE connection.

Then I installed my MOSQUITTO broker on a remote server controlled by me (with public static IP 2.XXX.XXX.XXX) and certificates generated by me through OpenSSL. The MQTT connection runs ok when using wifi connection (library PubSubClient) while the problem occurs when using LTE connection and SARA R5 modem.

I can say that same certificates work when using MQTT client as MQTTX or MQTT EXPLORER connecting to the same broker 2.XXX.XXX.XXX:8883.

I could verify that the source code written to set the modem and launch the MQTT connection works fine (result is always ok for each instruction). This is the code (which I had found as sample code around there):

```
setMQTTCommandCallback(mqttCallback);
...
setSecurityManager(SARA_R5_SEC_MANAGER_OPCODE_IMPORT, SARA_R5_SEC_MANAGER_ROOTCA, SEC_ROOT_CA, rootCa);
setSecurityManager(SARA_R5_SEC_MANAGER_OPCODE_IMPORT, SARA_R5_SEC_MANAGER_CLIENT_CERT, SEC_CLIENT_CERT, cert);
setSecurityManager(SARA_R5_SEC_MANAGER_OPCODE_IMPORT, SARA_R5_SEC_MANAGER_CLIENT_KEY, SEC_CLIENT_KEY, key);
resetSecurityProfile(LTE_SEC_PROFILE_MQTT);
configSecurityProfile(LTE_SEC_PROFILE_MQTT, SARA_R5_SEC_PROFILE_PARAM_CERT_VAL_LEVEL, SARA_R5_SEC_PROFILE_CERTVAL_OPCODE_YESNOURL);
configSecurityProfile(LTE_SEC_PROFILE_MQTT, SARA_R5_SEC_PROFILE_PARAM_TLS_VER, SARA_R5_SEC_PROFILE_TLS_OPCODE_VER1_2);
configSecurityProfile(LTE_SEC_PROFILE_MQTT, SARA_R5_SEC_PROFILE_PARAM_CYPHER_SUITE, SARA_R5_SEC_PROFILE_SUITE_OPCODE_PROPOSEDDEFAULT);
configSecurityProfileString(LTE_SEC_PROFILE_MQTT, SARA_R5_SEC_PROFILE_PARAM_ROOT_CA, SEC_ROOT_CA);
configSecurityProfileString(LTE_SEC_PROFILE_MQTT, SARA_R5_SEC_PROFILE_PARAM_CLIENT_CERT, SEC_CLIENT_CERT);
configSecurityProfileString(LTE_SEC_PROFILE_MQTT, SARA_R5_SEC_PROFILE_PARAM_CLIENT_KEY, SEC_CLIENT_KEY);
configSecurityProfileString(LTE_SEC_PROFILE_MQTT, SARA_R5_SEC_PROFILE_PARAM_SNI, broker);
setMQTTclientId(id);
setMQTTserver(broker, MQTT_BROKER_PORT);
setMQTTsecure(true, LTE_SEC_PROFILE_MQTT);
setMQTTcredentials(userName, passWord);
connectMQTT();

...

```
All above instructions return `SARA_R5_error_t=0` (success). But when calling `getMQTTprotocolError()` inside `mqttCallback()`, I got an error on command `SARA_R5_MQTT_COMMAND_LOGIN` (command=1) with `error codes 13 and 8`. This is the source code:

```
...
void mqttCallback(int command, int result) {
    if (result == 0) {
      int code1, code2;
      SARA_R5_error_t err = getMQTTprotocolError(&code1, &code2);
      if (SARA_R5_SUCCESS == err) {
        log_e("command %d protocol error code1 %d code2 %d", command, code1, code2);
      } else {
        log_e("command %d protocol error failed with error", command, err);
      }
    } 
...

```
This is the output:

`15:44:16.186 ---> mqttCallback(): command 1 protocol error code1 13 code2 8`

I tried also to comment the following line:

`configSecurityProfileString(LTE_SEC_PROFILE_MQTT, SARA_R5_SEC_PROFILE_PARAM_SNI, broker);`

and to disable client certificate evaluation:

`configSecurityProfile(LTE_SEC_PROFILE_MQTT, SARA_R5_SEC_PROFILE_PARAM_CERT_VAL_LEVEL, SARA_R5_SEC_PROFILE_CERTVAL_OPCODE_NO);
`

without success.

Any help would be very appreciated!

David

PaulZC, 2024-07-19:
Hi David,

All I can suggest is checking that you are passing the certs and key correctly formatted as ```String```.

I did run into an issue where I had to use [this exact formatting](https://github.com/sparkfun/SparkFun_RTK_EVK/blob/90e211f8a6db60857b0cb7d42caefcc9c30b4011/Example_Sketches/8_5_PointPerfect_MQTT/secrets.h#L30-L33) to define the certs and key in PROGMEM. IIRC, if I included a line break before the ```-----BEGIN```, it failed...

Code1 13 is the MQTT error class. Code2 8 is an "internal error"... Not much help...

Are there any more clues in the debug messages (```enableDebugging```)?

I hope this helps,
Paul

Davefries1966, 2024-07-19:
Thanks Paul for your prompt reply.

All I can say is that the same cert and key `String` variables are used by `PubSubClient` without any issue.

Is there an option on SARA R5 API to disable "SSL Secure" in the same way I can do on MQTTX or MQTT EXPLORER ?

Anyway I will try your suggested format.

David

Davefries1966, 2024-07-19:
Comparing the debugging obtained from both test connections, that is, test.mosquitto.org:8883 and my_broker:8883, I got in the first case (test.mosquitto.org):

```
20:35:54.014 -> bufferedPoll: start of event: +UUMQTTC: 1,1
20:35:54.014 -> processReadEvent: MQTT command result
20:35:54.014 -> [_][LTE.h:298] mqttCallback(): login
20:35:54.014 -> [_][LTE.h:631] setState(): state change 6(mqtt)
```

In the latter case (my MQTT broker):

```
20:32:51.347 -> bufferedPoll: start of event: +UUMQTTC: 1,0
20:32:51.381 -> processReadEvent: MQTT command result
20:32:51.381 -> sendCommandWithResponse: Command: +UMQTTER
20:32:51.417 -> sendCommandWithResponse: Response: 
20:32:51.417 -> +CSCON: 1
20:32:51.417 -> 
20:32:51.417 -> +UMQTTER: 13,8
```

It seems that the only difference is `+UUMQTTC: 1,0` instead of `+UUMQTTC: 1,1`

PaulZC, 2024-07-20:
Hi David,

```+UUMQTTC: 1,0``` is MQTT login fail.

Agreed, Code2 == 8 is Cannot set secure socket. (I was looking at an old version of the AT Manual previously.)

Maybe you need to change the ```configSecurityProfile(LTE_SEC_PROFILE_MQTT, SARA_R5_SEC_PROFILE_PARAM_TLS_VER, SARA_R5_SEC_PROFILE_TLS_OPCODE_VER1_2);``` ?

https://github.com/sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library/blob/cc961b2870059eb7a9f91cfc99af18187d614ae2/src/SparkFun_u-blox_SARA-R5_Arduino_Library.h#L575-L588

https://github.com/sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library/blob/cc961b2870059eb7a9f91cfc99af18187d614ae2/src/SparkFun_u-blox_SARA-R5_Arduino_Library.h#L598-L605

I hope this helps,
Paul

Davefries1966, 2024-07-22:
IT DOES WORK NOW! :-)

I re-made all the certificates and changed the setting of param `SARA_R5_SEC_PROFILE_PARAM_TLS_VER` to `SARA_R5_SEC_PROFILE_TLS_OPCODE_ANYVER`.

Thanks Paul again for your precious support! :-)

PaulZC, 2024-07-22:
No problem David - I'm glad it's working...

Closing...

Best wishes,
Paul

closed 2024-07-22 by PaulZC