sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library #51: Clarification about gpsPower() API

Davefries1966, 2024-08-07:
Here I am again with a new question regarding the Sparkfun AssetTracker board and the SARA-R5 Arduino Library.

I was wondering how to invoke the API function `gpsPower()`, and how to relate that function to SARA-R5 AT command `AT+UGPS`:

```
  // GPS
  typedef enum
  {
    GNSS_SYSTEM_GPS = 1,
    GNSS_SYSTEM_SBAS = 2,
    GNSS_SYSTEM_GALILEO = 4,
    GNSS_SYSTEM_BEIDOU = 8,
    GNSS_SYSTEM_IMES = 16,
    GNSS_SYSTEM_QZSS = 32,
    GNSS_SYSTEM_GLONASS = 64
  } gnss_system_t;
  typedef enum
  {
    GNSS_AIDING_MODE_NONE = 0,
    GNSS_AIDING_MODE_AUTOMATIC = 1,
    GNSS_AIDING_MODE_ASSISTNOW_OFFLINE = 2,
    GNSS_AIDING_MODE_ASSISTNOW_ONLINE = 4,
    GNSS_AIDING_MODE_ASSISTNOW_AUTONOMOUS = 8
  } gnss_aiding_mode_t;
  bool isGPSon(void);
  SARA_R5_error_t gpsPower(bool enable = true,
                           gnss_system_t gnss_sys = GNSS_SYSTEM_GPS,
                           gnss_aiding_mode_t gnss_aiding = GNSS_AIDING_MODE_AUTOMATIC);

```

**First question**: the call to `gpsPower()` function is mandatory ? Actually the board get GPS fixes also without calling that function.

**Second question**: as far as I understood, you cannot set the `gnss_sys` parameter with a bitmask of multiple values (while you can do that with the corresponding AT command) because the API function only accepts an enumerated value. Then I tried to invoke successively multiple instances of the `gpsPower() `function, each instance with different parameter, for example:

```
      gpsPower(true, GNSS_SYSTEM_GPS); delay(100);
      gpsPower(true, GNSS_SYSTEM_GALILEO); delay(100);
      gpsPower(true, GNSS_SYSTEM_BEIDOU); delay(100);
      gpsPower(true, GNSS_SYSTEM_GLONASS); delay(100);
```

But, the AT debugging says that GPS is the only enabled GNNS system:

```
08:59:06.702 -> AT+UGPS?
08:59:06.702 -> 
08:59:06.702 -> +UGPS: 1,1,1
```

On the other hand, you can set bitmasks through AT command "AT+UGPS", for example the AT command:

`AT+UGPS=1,1,77`

should set `aiding_mode=1` (=automatic local aiding) and `GNSS_systems=77` (=GPS+GLONASS+BEIDOU+GALILEO).

In this specific case, the strange behaviour I have found is that the `AT+UGPS?` command reports:

```
09:35:18.913 -> AT+UGPS=1,1,77
09:35:18.913 -> AT+UGRMC=1

...

09:36:07.109 -> AT+UGPS?
09:36:07.145 -> 
09:36:07.145 -> +UGPS: 1,1,99
```

that is, `99` instead of `77` !

This is the code (is it correct?):

```
      enableAtDebugging();

      enableGNSSAntennaPower();
      setUtimeConfiguration();  // Use default offset (offsetNanoseconds = 0, offsetSeconds = 0)

      const char* str2 = "=1,1,77";
      char atCommand[100];
      strcpy(atCommand, SARA_R5_GNSS_POWER);
      strcat(atCommand, str2);
      sendCommand(atCommand, true);

      gpsEnableRmc();  // Enable GPRMC messages
      setGpsReadCallback(&processGpsReadStatic);
```

Any help would be very much appreciated! :-)

David

PaulZC, 2024-08-07:
Hi David,

Just a quick reply:

There are some notes here:

https://github.com/sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library/blob/cc961b2870059eb7a9f91cfc99af18187d614ae2/examples/SARA-R5_Example1_GNSS_GPRMC/SARA-R5_Example1_GNSS_GPRMC.ino#L87-L88

You should be able to OR the GNSS Systems. Please try:

```
gpsPower(true, GNSS_SYSTEM_GPS | GNSS_SYSTEM_GALILEO | GNSS_SYSTEM_GLONASS);
```

It is an M8 GNSS module, so it can only receive three GNSS simultaneously:

[image]

Best,
Paul

Davefries1966, 2024-08-07:
Ok thanks Paul.. so a correct parameter could be for example 69 (GPS+GLONASS+GALILEO).

I didn't understand if the function call to `gpsPower() `is mandatory or not, and even where this function call must be placed in the code. I would mean if the function call must be invoked only once at initial phase:

```
...
      enableGNSSAntennaPower();
      gpsPower(true, GNSS_SYSTEM_GPS | GNSS_SYSTEM_GALILEO | GNSS_SYSTEM_GLONASS);
      gpsEnableRmc();  // Enable GPRMC messages
      setGpsReadCallback(&processGpsReadStatic);
...
```
or, each time a gps request is performed in the code loop:

```
...
     if (gpsRequest(GNSS_REQUEST_TIMEOUT, GNSS_REQUEST_ACCURACY) == SARA_R5_SUCCESS) {
      <some code here>
     } 
...
```

because my feeling is that GPS power turns off after a while (seen `AT+UGPS=0` and `+UGPS? 0` on the AT debugging), not being sure where or when it is switched off.

**Beyond all, my goal is to have the maximum GPS coverage and speed up as much as possible the "time-to-first-fix" without using online services.**

PaulZC, 2024-08-08:
Hi David,

Apologies for the slow reply.

I don't have a lot of experience with the SARA-R5 GNSS, but I do remember it being 'tricky' when I was writing the library and code examples. The main one being [AssetTracker Example 23](https://github.com/sparkfun/MicroMod_Asset_Tracker/blob/main/Examples/AssetTracker_Example23_GNSS_Request/AssetTracker_Example23_GNSS_Request.ino)

I left notes to myself in the code:

https://github.com/sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library/blob/cc961b2870059eb7a9f91cfc99af18187d614ae2/src/SparkFun_u-blox_SARA-R5_Arduino_Library.cpp#L5238-L5242

https://github.com/sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library/blob/cc961b2870059eb7a9f91cfc99af18187d614ae2/src/SparkFun_u-blox_SARA-R5_Arduino_Library.cpp#L5153-L5162

It does seem slightly crazy to have to turn the GPS power _off_ before requesting the location. But that is what I found at the time. Maybe I just didn't get that bit right...?

```gpsRequest``` uses the SARA_R5_GNSS_REQUEST_LOCATION "+ULOC" command to ask for hybrid localization information. It requests a "2: single shot position". The "sensor" defaults to "3" which is "1: use the GNSS receiver for localization" + "2: use cellular CellLocate® location information".

For your application, I don't think you're interested in the hybrid location - with CellLocate. I think the standard NMEA RMC (Recommended Minimum) is enough? So we should concentrate on that. SARA_R5_GNSS_GPRMC "+UGRMC" is enabled by ```gpsEnableRmc```, and read by ```gpsGetRmc```. ```gpsGetRmc``` parses the position, speed and clock data into three data structures. ```gpsGetRmc``` is used in [AssetTracker Example 11](https://github.com/sparkfun/MicroMod_Asset_Tracker/blob/main/Examples/AssetTracker_Example11_GNSS_GPRMC/AssetTracker_Example11_GNSS_GPRMC.ino).

In Example 11, I don't call ```gpsPower``` because ```setUtimeMode``` ("+UTIME") enables the GPS module. Enabling the power is mandatory, but "+UTIME=1,1", does it automatically.

If you don't need the Pulse-Per-Second signal, maybe it is beneficial to call ```gpsPower``` instead of ```setUtimeMode```.

In your case, maybe you should disable the automatic local aiding too? So, the call would be:

```
gpsPower(true, GNSS_SYSTEM_GPS + GNSS_SYSTEM_SBAS + GNSS_SYSTEM_GALILEO + GNSS_SYSTEM_GLONASS, 0);
```

Remember you can only enable three GNSS. You get SBAS for free with GPS. QZSS is also free, but is mainly for Japan.

The GPS Read Callback only applies to the "+UULOC" hybrid location unsolicited message. For your application, you should not call ```setGpsReadCallback```.

I hope this helps.

When I get some time, I should add support for "+UGNMEA". That would allow NMEA messages to be output as URC unsolicited messages and given their own callbacks... A much better solution than getting the messages...

All the best,
Paul

Davefries1966, 2024-08-08:
Hi Paul,

now the topic is much more clear to me. :-)

I'll have to re-arrange the my source code with getting NMEA messages instead of using request/callback mechanism, which would be preferrable in order to detach the request from the reply (where in my case I also send the fix through MQTT...).

Cheers
David

PaulZC, 2024-08-12:
Closing... Please reopen if you need more help with this.

You are welcome to open a new "Feature Request" issue - requesting support for +UGNMEA.

Best,
Paul

closed 2024-08-12 by PaulZC

Davefries1966, 2024-08-12:
Thank you Paul. 

Actually it was easy to create a second timed task handling `gpsGetRmc` requests without interfering with the main task. It's working now in PPS mode (i.e. using function `setUtimeMode`) because that seems more reliable: tried also using `gpsPower` with various GNSS combinations (maximum 3 at a time) but always got `SARA_R5_ERROR_UNEXPECTED_RESPONSE` (error code =4) even under a clear sky. The strange is that yesterday evening it was working flawlessly with `gpsPower` (instead of `setUtimeMode`), GPS fixes were even faster with respect to PPS mode. Today something is working wrong... anyway PPS mode is fine for me, positions are also very precise. It takes about 10 seconds to get first valid RMC. 

P.S. the "best effort" approach, that is a combinatoin of GNSS + LTE, applies only to PPS mode ? When setting `gpsPower` with for example GPS+GNSS+GALILEO, it will use also LTE ?  

Cheers
David