sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library #52: Unable to connect the SARA R510M8S module to the network with Hologram SIM, despite correct APN configuration in Vietnam.

zumintam, 2024-11-06:
Your workbench
Platform: Development board or microcontroller with SARA R510M8S module.
Device version: SARA R510M8S module.
Firmware version: Ensure you are using the latest firmware version for compatibility.
Wiring: The SARA R510M8S module is connected to the microcontroller via UART or other communication interfaces.
Power: Powered by an appropriate DC source (typically 5V for SARA R5 with V_ext).
Additional details: Hologram SIM is used, and the APN "hologram" is configured correctly, but the device cannot find the network or connect.

PaulZC, 2024-11-06:
Hi @zumintam ,

According to https://www.hologram.io/coverage/ , the providers for Vietnam are:

[image]

Is your area covered by MobiFone or Vinaphone? Do they provide **LTE-M** coverage in your area?

I hope this helps,
Paul

zumintam, 2024-11-07:
[SARA-R510M8S-00B-01 Registration Status Not Registered(1).pdf](https://github.com/user-attachments/files/17656135/SARA-R510M8S-00B-01.Registration.Status.Not.Registered.1.pdf)

zumintam, 2024-11-07:
I noticed that several other people are experiencing the same issue as I am, as seen in this forum, and there has still been no response 
https://community.sparkfun.com/t/sara-r5-registration-status-not-registered/45663/6

zumintam, 2024-11-08:
Dear Paul,

Thank you for the instructions. I will proceed with the update as you suggested. 
I will let you know if any further issues arise. Thank you for your support!

Best regards,
Tam Vu

zumintam, 2024-11-08:
After reading this, do you think Vietnam's network infrastructure is sufficient for supporting LTE-M effectively?
[Dear Paul.pdf](https://github.com/user-attachments/files/17674780/Dear.Paul.pdf)

PaulZC, 2024-11-08:
Hello Tam Vu,

It looks like there are no LTE-M operators in your area.

In my area of the UK, I see only:

[image]

But with the LARA-R6001D and the same SIM, I see:

[image]

Would you like us to ask u-blox for suggestions?

Best wishes,
Paul

zumintam, 2024-11-27:
**Subject:** Inquiry About Bulk Pricing for SparkFun GPS-RTK2 Board (ZED-F9P)  

Dear SparkFun Team,  

I hope this email finds you well.  

I am interested in purchasing the **SparkFun GPS-RTK2 Board - ZED-F9P (Qwiic)** (GPS-15136) for a project in Vietnam. The listed price is $249.95, and I was wondering if you could offer a better price.  

Could you also let me know the minimum quantity required to qualify for bulk pricing and the best price you can offer at different quantity tiers? Additionally, I would appreciate information about shipping options and costs to Vietnam.  

Thank you for your support, and I look forward to your response.  

Best regards,  
Tam Vu
vuminhtamndc@gmail.com

PaulZC, 2024-11-28:
Hello Tam Vu,

I will pass your message to my colleagues. SparkFun is closed for the Thanksgiving holiday - it will take them a few days to reply.

Best wishes,
Paul

closed 2024-11-28 by PaulZC