sparkfun/SparkFun_u-blox_SARA-R5_Arduino_Library #56: SARA R5 clock() function always starts from "15/01/01,00:00:00+00"

Davefries1966, 2025-04-01:
Hi all,
I wish to report a new issue regarding the use of the SparkFun Asset Tracker board (new sample) with an ESP32 processor in the Arduino development environment. When calling the **clock()** function from the SARA R5 library, I noticed that the time always starts from January 1, 2015, at 00:00:00 UTC, even though the RMC fixes show the correct time (today being April 1, 2025, etc.). Please note that I am using the board in Wi-Fi mode, without a SIM card. I also tried setting the clock using the m-center software, but the board does not retain the setting; every time it is powered off and on again, the time resets to January 1, 2015.

Thank you in advance for your support!

David Frittella

PaulZC, 2025-04-01:
Hi David (@Davefries1966 ),

I think this is possibly expected behaviour? If you disconnect the power, the SARA-R5 has no way to maintain its internal clock.

Are you asking if it is possible to sync the ```clock()``` from GNSS time (RMC)?

Best wishes,
Paul

Davefries1966, 2025-04-02:
Thank you, Paul, for your immediate feedback. This is the second unit of the Asset Tracker board that I am using. I am quite sure that on the first unit, the `clock()` function returned a time consistent with the GNSS fix. The only difference is that on the first unit, I had also activated the LTE SIM.

Anyway, the answer is yes, I would like to receive the GNSS time in the `clock()` function. Would it be possible?

Cheers,
David

Davefries1966, 2025-04-04:
Solved ! :-)

It can be closed.

Cheers
David

closed 2025-04-04 by PaulZC

Davefries1966, 2025-04-04:
Sorry Paul, you are right. The solution was as suggested, that is, set clock (once) by the first valid GNSS fix:

```
  bool clockSet = false;

  if (valid) { // valid GNSS fix
    if (!clockSet) {
      setClock(clck.date.year, clck.date.month, clck.date.day, clck.time.hour, clck.time.minute, clck.time.second, 0);
      clockSet = true;
    }

  ...
}
```

Cheers and thanks again! :-)
David